MickMake/GoSungrow #116: Cannot connect to european server

thesnable, 2024-06-03:
https://www.isolarcloud.eu

wont connect ... :(

Paraphraser, 2024-06-04:
If you want help, I think you'll have to provide a *little* more information - such as:

* Are you running GoSungrow in Home Assistant? If yes then:

	- Settings » Add-ons » GoSungrow » Configuration tab, then open the page menu (the &#x22EE; near the top, right) and choose "Edit in YAML". Copy and paste what you see into a text editor, sanitise anything you shouldn't be sharing (`sungrow_user`, `sungrow_password`, `mqtt_user` and `mqtt_password`) and include it in your reply, wrapped in code fences (see further down for an explanation). Here's mine:

		```
		sungrow_host: https://augateway.isolarcloud.com
		sungrow_user: aabbccddee
		sungrow_password: secrets
		sungrow_appkey: B0455FBE7AA0328DB57B59AA729F05D8
		mqtt_host: core-mosquitto
		mqtt_port: "1883"
		mqtt_user: moi
		mqtt_password: secrets
		debug: true
		sungrow_timeout: 60
		```
		
	- Settings » Add-ons » GoSungrow » Log tab, and then copy/paste the lines showing the error message.

* Alternatively, if you are running GoSungrow from the command line:

	- Run `GoSungrow config read`, Copy and paste what you see into a text editor, sanitise anything you shouldn't be sharing (same as before, `--user`, `--password`, `--mqtt-user` and `--modbus-password`) and include it in your reply, wrapped in code fences. Here's mine: 

		```
		+-------------------+------------+---------------------------+--------------------------------+---------------------------------------+
		|       FLAG        | SHORT FLAG |        ENVIRONMENT        |          DESCRIPTION           |          VALUE (* = DEFAULT)          |
		+-------------------+------------+---------------------------+--------------------------------+---------------------------------------+
		| --config          |            | GOSUNGROW_CONFIG          | GoSungrow: config file.        | /Users/moi/.GoSungrow/config.json |
		| --debug           |            | GOSUNGROW_DEBUG           | GoSungrow: Debug mode.         | false *                               |
		| --quiet           |            | GOSUNGROW_QUIET           | GoSungrow: Silence all         | false *                               |
		|                   |            |                           | messages.                      |                                       |
		| --timeout         |            | GOSUNGROW_TIMEOUT         | Web timeout.                   | 30s *                                 |
		| --user            | -u         | GOSUNGROW_USER            | SunGrow: api username.         | someone@domain.com                 |
		| --password        | -p         | GOSUNGROW_PASSWORD        | SunGrow: api password.         | secrets                    |
		| --appkey          |            | GOSUNGROW_APPKEY          | SunGrow: api application key.  | B0455FBE7AA0328DB57B59AA729F05D8      |
		| --host            |            | GOSUNGROW_HOST            | SunGrow: Provider API URL.     | https://augateway.isolarcloud.com     |
		|                   |            |                           |                                | *                                     |
		| --token-expiry    |            | GOSUNGROW_TOKEN_EXPIRY    | SunGrow: last login.           | 2024-05-02T08:54:25                   |
		| --save            | -s         | GOSUNGROW_SAVE            | Save output as a file.         | false *                               |
		| --dir             |            | GOSUNGROW_DIR             | Save output base directory.    |  *                                    |
		| --mqtt-user       |            | GOSUNGROW_MQTT_USER       | HASSIO: mqtt username.         |  *                                    |
		| --mqtt-password   |            | GOSUNGROW_MQTT_PASSWORD   | HASSIO: mqtt password.         |  *                                    |
		| --mqtt-host       |            | GOSUNGROW_MQTT_HOST       | HASSIO: mqtt host.             |  *                                    |
		| --mqtt-port       |            | GOSUNGROW_MQTT_PORT       | HASSIO: mqtt port.             |  *                                    |
		| --modbus-user     |            | GOSUNGROW_MODBUS_USER     | Modbus username.               |  *                                    |
		| --modbus-password |            | GOSUNGROW_MODBUS_PASSWORD | Modbus password.               |  *                                    |
		| --modbus-host     |            | GOSUNGROW_MODBUS_HOST     | Modbus host.                   |  *                                    |
		| --modbus-port     |            | GOSUNGROW_MODBUS_PORT     | Modbus port.                   | 502 *                                 |
		+-------------------+------------+---------------------------+--------------------------------+---------------------------------------+
		```
		
	- Run the command that is failing and include the lines that show the error, wrapped in code fences.

If you are unfamiliar with the term "code fences", it means lines containing triple back-ticks, like this:

\`\`\`

your lines here

\`\`\`

Code fences stop MarkDown from trying to run lines together to form paragraphs and allow you (the author) to be able to show recipients precisely what you're seeing.

thesnable, 2024-06-09:
Hi!

Well sorry .. yes :) HA!

sungrow_host: https://www.isolarcloud.eu/
sungrow_user: xxx
sungrow_password: xxx
sungrow_appkey: 93D72E60331ABDCDC7B39ADC2D1F32B3
mqtt_host: core-mosquitto
mqtt_port: "1883"
mqtt_user: shatner
mqtt_password: shatner
debug: false
sungrow_timeout: 60

The Output i get:
[14:54:30] INFO: Login to iSolarCloud using gateway https://www.isolarcloud.eu/ ...
Error: API httpResponse is 404 Not Found
Usage:
  GoSungrow api login [flags]

Examples:
	GoSungrow api login  

Flags: Use "GoSungrow help flags" for more info.

Additional help topics:

ERROR: API httpResponse is 404 Not Found
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

When i use the redirect server after login (isolar server 
[14:56:42] INFO: Login to iSolarCloud using gateway https://web3.isolarcloud.eu/ ...
Error: API httpResponse is 405 Not Allowed
Usage:
  GoSungrow api login [flags]

Examples:
	GoSungrow api login  

Flags: Use "GoSungrow help flags" for more info.

Additional help topics:

ERROR: API httpResponse is 405 Not Allowed
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

thesnable, 2024-06-09:
Hi!

Well sorry .. yes :) HA!

sungrow_host: https://www.isolarcloud.eu/
sungrow_user: xxx
sungrow_password: xxx
sungrow_appkey: 93D72E60331ABDCDC7B39ADC2D1F32B3
mqtt_host: core-mosquitto
mqtt_port: "1883"
mqtt_user: xxx
mqtt_password: xxx
debug: false
sungrow_timeout: 60

The Output i get:
[14:54:30] INFO: Login to iSolarCloud using gateway https://www.isolarcloud.eu/ ...
Error: API httpResponse is 404 Not Found
Usage:
GoSungrow api login [flags]

Examples:
GoSungrow api login

Flags: Use "GoSungrow help flags" for more info.

Additional help topics:

ERROR: API httpResponse is 404 Not Found
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

When i use the redirect server after login (isolar server
[14:56:42] INFO: Login to iSolarCloud using gateway https://web3.isolarcloud.eu/ ...
Error: API httpResponse is 405 Not Allowed
Usage:
GoSungrow api login [flags]

Examples:
GoSungrow api login

Flags: Use "GoSungrow help flags" for more info.

Additional help topics:

ERROR: API httpResponse is 405 Not Allowed
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

Paraphraser, 2024-06-09:
The two most immediate problems are in these lines:

```
sungrow_host: https://www.isolarcloud.eu/
sungrow_appkey: 93D72E60331ABDCDC7B39ADC2D1F32B3
```

Assuming the EU server is what you need, then that URL should be:

```
https://gateway.isolarcloud.eu
```

and the APPkey will be one of these:

* `B0455FBE7AA0328DB57B59AA729F05D8`
* `ANDROIDE13EC118BD7892FE7AB5A3F20`

However, **before** you rush off and try that and find that it still doesn't work, can I please suggest that you at least read through [Part 2 of this gist](https://gist.github.com/Paraphraser/cad3b0aa6428c58ee87bc835ac12ed37#encryptionhack).

If you start to read the gist and think, "well, I've done all this already" then that's cool. It just means you missed the [configuration step](https://gist.github.com/Paraphraser/cad3b0aa6428c58ee87bc835ac12ed37#configuration), which is the information about the URL and APPKey above.

Alternatively, if you start to read the gist and it doesn't sound familiar then you need to follow **all** the steps in part 2.

Hope this helps.

thesnable, 2024-06-12:
hello, thank you very much for your assistance. I made it work. I think the AP codes and the wrong gateway was the main reason. I'm sorry I didn't fidget it out the first way I thought that I need unique or different AP keys, but I did not find them, sorry for that. Sorry for that. Finally, I ended up using Home Assistant mod bus connection. Internet Values are often hallucinations 😂✅

thanks