MickMake/GoSungrow #118: Stopped working without any changes.

clintebbesen, 2024-07-08:
[01:59:46] INFO: Login to iSolarCloud using gateway https://augateway.isolarcloud.com ...
Error: unknown error 'Request is not encrypted'
Usage:
  GoSungrow api login [flags]

Examples:
	GoSungrow api login  

Flags: Use "GoSungrow help flags" for more info.

Additional help topics:

ERROR: unknown error 'Request is not encrypted'
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

Paraphraser, 2024-07-08:
I'm going to assume that you are trying to use GoSungrow from the command line.

> It doesn't actually matter if the error log is from a copy-and-paste from the GoSungrow add-on in Home Assistant. You just need to follow different steps.

What you need to do is do a quick skim-read of [this gist](https://gist.github.com/Paraphraser/cad3b0aa6428c58ee87bc835ac12ed37).

If you are happy to recompile GoSungrow from its source code then you will need to follow [Part 1 of the gist](https://gist.github.com/Paraphraser/cad3b0aa6428c58ee87bc835ac12ed37#part1).

If recompiling is something you don't want to tackle then you will need to follow [Part 3 of the gist](https://gist.github.com/Paraphraser/cad3b0aa6428c58ee87bc835ac12ed37#part3).

Irrespective of the approach you use, one common mistake I've noticed other people making is not realising that all four parts of the gist finish up by directing you to the section on [Configuration](https://gist.github.com/Paraphraser/cad3b0aa6428c58ee87bc835ac12ed37#configuration).

It's also important to realise the configuration of both the server field and App Key is not an exact science. You might have to try various combinations. My guess is that, if GoSungrow was working before then the host will already be correct and you'll probably only need to futz with the App Key.

Hope this helps.

clintebbesen, 2024-07-08:
I installed it via add ons in home assistant. It has been running and working great for about 2 months and last week stopped working, when checking logs under the add on in home assistant this is what I found. I haven’t made any changes to config or anything.

Paraphraser, 2024-07-08:
"Request is not encrypted" is characteristic of the unpatched version of GoSungrow.

Other than to express surprise that it's only just happened, I can't speak to the timing. There are a lot of things about the iSolarCloud side of things which don't really make a whole lot of sense (eg why some people need one of the API  keys, other people the other key, and then find that they need to switch keys for no apparent reason).

**But** what you will have to do is follow all the steps in Part 2 of the gist I linked you to earlier.

Paraphraser, 2024-07-08:
And, just to be clear, **everyone** using GoSungrow (command line, add-on, container) can claim, accurately and truthfully, that it goes nuts despite nothing changing at the client end. All these problems are triggered by changes on the server side (iSolarCloud).

What needs to be kept in mind is that Sungrow hasn't published an API and neither have they guaranteed any kind of reliability. We users are all firmly in "hoping for the best" mode.

clintebbesen, 2024-07-08:
Awesome thank you! I’ll have a look and see what I can do. 

If it means anything, around the time that it stopped working, the iOS isolarcloud app had asked me to reauthenticate so maybe something has changed in their app.

Paraphraser, 2024-07-08:
Every time that iPad app asks me to re-authenticate, I hold my breath until I've gone to a Mac and done a `GoSungrow api login` because, like you, I'm assuming that could be a trigger for the app key switch. Touch wood - hasn't happened yet.

One thing I have **never** done is login to iSolarCloud via the web...

clintebbesen, 2024-07-08:
Awesome thank you! 

I did those steps in the gist after I initially set it up and it wasn't working, doing that got it working. I went through and did the same steps now, didnt change the config, clicked start on the add on in home assistant and its now working again. Thank you! who knows why I had to do step 2 on the gist again, but it worked.

11notes, 2024-09-25:
You can try my latest built of this code with some added improvements: https://hub.docker.com/r/11notes/sungrow