MickMake/GoSungrow #119: Unable to display Energy data

matthewbayard, 2024-07-19:
I have been able to pull all the sungrow data into HA. All data like energy, loads ect are working.

However I can't add the correct ones into Energy section.

I get this error 

Last reset missing
The following entities have state class 'measurement' but 'last_reset' is missing:
sensor.gosungrow_virtual_1228101_7_1_1_p8063
sensor.gosungrow_getpsdetail_1228101_p83102

[image]

Any help with finding the correct sensor ID to use?

thanks

Robin19777, 2024-11-28:
copied from https://github.com/MickMake/HomeAssistantAddons/blob/main/GoSungrow/DOCS.md

HA's Energy Dashboard
To use HA's Energy Dashboard, you'll need to find the following entries. Make sure to replace XXXXXXXXXXXXXX with your ps_key.

Electricity grid
sensor.gosungrow_virtual_XXXXXXXXXXXXXX_grid_to_load_energy
sensor.gosungrow_virtual_XXXXXXXXXXXXXX_pv_to_grid_energy
Solar Panels
sensor.gosungrow_virtual_XXXXXXXXXXXXXX_pv_energy
Home Battery Storage
sensor.gosungrow_virtual_XXXXXXXXXXXXXX_battery_discharge_energy
sensor.gosungrow_virtual_XXXXXXXXXXXXXX_battery_charge_energy

bamcauliffe, 2025-04-09:
I am having a similar problem.
I did find that information in the Add-on Documentations.

I have not been able to find these entities, so I am currently using

Electricity grid
sensor.gosungrow_virtual_1202522_7_1_1_p8063
sensor.gosungrow_virtual_1202522_7_1_1_p8062

Solar Panels
sensor.gosungrow_virtual_devices_xxxxxxxxx_today_energy

However these entitles are not state_class: total_increasing, with a correct last_reset date/time.

If  I can not find the entities 
Electricity grid
sensor.gosungrow_virtual_XXXXXXXXXXXXXX_grid_to_load_energy
sensor.gosungrow_virtual_XXXXXXXXXXXXXX_pv_to_grid_energy

Solar Panels
sensor.gosungrow_virtual_XXXXXXXXXXXXXX_pv_energy

Home Battery Storage
sensor.gosungrow_virtual_XXXXXXXXXXXXXX_battery_discharge_energy
sensor.gosungrow_virtual_XXXXXXXXXXXXXX_battery_charge_energy

Does it mean that these are not in the version of the Add on that I have installed V3.0.7?
Do I need to remove and re-install the Add-on?
Does the firmware of the Inverter need to be updated?

What do I need to do, to get these entities added to my system?

Do you have these entities? What version of the Add-on are you using?