MickMake/GoSungrow #122: Can't find any entities

maomao54922, 2024-10-02:
Hi all,

I'm very new to HA, currently using HAOS on VM. I think I have my Gosungrow running using the appkey ANDROIDE13EC118BD7892FE7AB5A3F20. My log shows it is syncing data from gateway. However i can't get the add-on to sidebar, nor finding any entities execpt the update one. Can anyone tell me if I've missed something?

`[07:52:57] INFO: Login to iSolarCloud using gateway https://augateway.isolarcloud.com ...
Logging in again
Email:	FFionleesinyee@hotmail.com
Create Date:	Tue Mar 12 15:25:53 CST 2024
Login Last Date:	2024-10-02 15:52:58
Login Last IP:	
Login State:	1
User Account:	
User Id:	
User Name:	
Is Online:	false
Token:	488629_e54c15ba70fe4e0d98e3b330eab7de28
Token File:	/data/.GoSungrow/AppService_login.json
[07:52:58] INFO: Syncing data from gateway https://augateway.isolarcloud.com ...
2024/10/02 07:52:58 INFO: Connecting to MQTT HASSIO Service...
2024/10/02 07:52:58 INFO: Connecting to SunGrow...
2024/10/02 07:52:58 INFO: Found SunGrow 3 devices
2024/10/02 07:52:58 INFO: Caching Sungrow metadata...
2024/10/02 07:52:58 INFO: Cached 484 Sungrow data points...
2024/10/02 07:52:59 INFO: Syncing 2942 entries with HASSIO from queryDeviceList.
2024/10/02 07:52:59 INFO: Syncing 148 entries with HASSIO from getPsList.
-CUCUCUCU--CUCUCUCU-CUCUCU--CUCUCU-CUCUCU-CU-CU-CU-CUCU--CU--CU-CUCU-CU-CUCUCU-CU--CU-CUCUCUCUCUCUCU---CU-CUCUCU-CU-CUCUCU-CUCU-CU-CU---CUCU-CU-CUCUCU-CU-CUCU-CUCUCU--CUCUCUCU--CUCU--CU----CU---CUCUCU--CUCUCUCUCUCUCU-CUCUCU-CUCU--CU-CUCUCUCU------CU-CUCUCUCUCU-CUCUCUCUCUCU--------CU-CUCUCUCUCU-CUCUCUCUCUCUCU-------CU-----CUCUCUCU---CU-CUCUCU-CU-CUCU-CUCUCUCUCUCUCUCUCUCU-CUCUCU-CU-CU-CU-CU-CUCUCUCU--CU-CUCU-CU--CUCUCU----CUCUCU-CUCUCUCU-CUCUCU--CU-CU-CUCUCU-----CU
CUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCU?CUCUCUCUCUCUCUCUCUCUCU?CUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCU?CUCUCUCUCUCUCUCUCUCUCUCUCUCU?CUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCU?CUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCU
2024/10/02 07:52:59 INFO: Syncing 246 entries with HASSIO from getPsDetail.
CUCUCU?CUCUCUCUCUCU?CUCUCUCUCUCU?CUCUCU?CUCUCUCUCUCUCUCUCUCUCUCU??CUCUCUCUCU?CUCUCUCUCU?CUCUCUCUCUCUCUCUCUCUCU?CUCUCUCUCUCUCUCUCUCUCUCU?CUCUCUCUCUCUCUCUCUCUCU?CU?CU?CUCUCUCU?CUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCU?CUCUCUCUCUCUCUCUCUCUCU?CUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCU?CUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCU?CUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCUCU?CUCUCUCUCUCUCUCUCUCUCU?CUCUCUCUCUCUCUCUCUCUCUCUCU
2024/10/02 07:52:59 INFO: Starting ticker...
2024/10/02 07:52:59 INFO: Fetch Schedule: 5m
2024/10/02 07:52:59 INFO: Sleep Delay:    40s
2024/10/02 07:59:01 INFO: Syncing 246 entries with HASSIO from getPsDetail.
2024/10/02 07:59:01 INFO: Syncing 2942 entries with HASSIO from queryDeviceList.
UUUUUUUUUUUUUUUUUUU??UU?UUUUUU?UCUUUCUCUUUCUUCUCUUUUCUUUUUUUU?UUUUUUUUUUUUUUUCUUUUUUU?UUUUU?CUUUUUUUUUUUUUUCUUUUU?UU??U?UUUUUUU?CUUUUUCUUCUCUUUCUUUUUCUUUUUCUUUUUUCUCUUUUUUCUUUUU?UUCUUU?UUUUUUUUUUUUUUUCUUUUUUUCUUUUUCUU?U?UUUUUUUUUUUUUUUCUUCUUUUUUUUUU?UUCUCUUU???CUUUUUUUUUCUUUU
UUCU-CUUUUU--CUUUCUUCU-UCU-U-CUUCU---UUUU-U-UUU-U-UUCU-CUU-UUUU-U--U-UCU-----U-U-UU-U-CUU-U-U-CU--U---UUUU-UUUCUUU-UU-CUU-CUU---U-UUCU-UUCU----UUCUUU-UU--CUCU-CUU-UUCU---CU-UUUUUU--CUCUU---U-CU----UCU-----UCU--UUUUUU-U-U--U-U--CUU-UUUCUU-CUCUUUCU-CU----CUCUUUCUUUCU-CUUUU--CUCU-UU----U-CU-UCU------CUU-UUUU-CU--UUU-CU----CU----CUCUUUUCUUCUU-UU----
2024/10/02 07:59:01 INFO: Syncing 148 entries with HASSIO from getPsList.
UUU??UUUUUUUUU?UUUCUUUUUUUUCUUU?UUUUUUUUUUUUUUUUUUUUUCUUUUUUUUUUUCUUUUUUCUUUCUUUCUUUUUUUUCUUUUUUUCUUUUUUUUU?UUUUUUUUUUUUCUUUUUUUUUUUUUUUUCUUUUUUCUUUCUUUUUUUUUUUU
2024/10/02 08:04:51 INFO: Syncing 2942 entries with HASSIO from queryDeviceList.
2024/10/02 08:04:51 INFO: Syncing 148 entries with HASSIO from getPsList.
CUCU--U-UU---CU--UCUCU-CU------CU--U-UUUCUUUCU--UCU-UCUU-UUUCUU--UCUUCUUUU--U---CUCUUUU-UU----UU-U-U-U--U---UCU--CU---U-CUCU-CUCUU-CU----U-CU-U----UCUUUUUUUUU--UUUUCU--UUCUU-CUUCU--U--CUCUUCUUUCUU-CUCU-CUUUUU-UU-CUUCUCUUU--CUCUUU---CUCUU--CUCU-U---CU--CUUCU--CU-CUCU---U-U----UU-CUCUCU-U-CUUCUCUCUCUUCUU--U-U-CU-CU-CUCUCUUUUUCU-U-----UUU-UCUU---CU--UUU-UCU--CU-CU-UU
UU?UUUUUUUUUUUUUUUUUCUUUUUUUUUUUUUUCUUUUCUUUCUU?UCUUUUUUUCUUUUU?UUCUUUUCUUUUUUUUUUUCUUCUUUUCU?UUUUUUUUUUUUUUUUUUUUUUCUUUUUUCUUUUUCUUUUUUUUUUUUUUU?UUUUUUUUUUUCUUUUU
2024/10/02 08:04:51 INFO: Syncing 246 entries with HASSIO from getPsDetail.
UUUUUU?UCUUUU?UUUUUUUUUUUUU?CU?UCUUUUCUUUUUUCUCUUCUCUUUUUUUU??UUUUUUUUUCUUUU?UUUUUUUU?UUU?UUUUCUUUUUUUCUCUCUUUCUUUUUUUUUUUU?CUUCUUUUUUUCUUCU?UUCUCUUU?UUUCUUUUUUUUUUUUUUUUUUUUUUUUUCUUUUCUUU?CUU?U?UUUUUCUU??UCUUUUUUUUUUUUUUCUUUUUUUUUCUCUUCUCUUCUUUUCUUUUUUUCUUU??UUUUUUUUUUUUCUUU?UUUU`

syerad, 2024-12-18:
I'm struggling with the same problem. Were you able to find out how see all the sensors?

bamcauliffe, 2025-04-09:
Maomao54922 & Syerad,
I am using the same appkey.
When I look at the GoSungrow logs, I see similar information where there is lots of "U" etc.
However I am getting lots of entities with data.

To see the data, I go to 
Settings
Devices & Services
MQTT (integration)
In there I have 6 x "devices" relating to GoSungrow including ".... meter", ".... Options", "..... inverter"

Or you can go to 
Settings
Devices & Services
Entities
The in the search field start typing "gosungrow".
I have over 100 entities.

My problem is getting the right entities to the HA Energy dashboard.
At the bottom of the Addin it does say which entities to use but I have not found these.
I will be raising this problem on a different issue here.