MickMake/GoSungrow #124: Sensor for live feed back to the grid

Hey mate I have it all working pretty well but can not find the sensor for live feed back to grid please help

keensy, 2024-12-17:


keensy, 2024-12-18:
I wanted to set up the energy app but some of the sensors etc didn't meet the purpose so I created these in my configuration.yaml hope this helps someone else

# Template Sensors
template:
  - sensor:
      # Solar Data - Anonymized Total
      - name: "Total Solar Output"
        unique_id: "total_solar_output"
        unit_of_measurement: "kWh"
        state_class: "measurement"
        device_class: "energy"
        state: >-
          {% set solar_value = states('sensor.gosungrow_virtual_1134569_1_1_1_p24') | float(0) %}
          {{ solar_value if solar_value > 0 else 0 }}

      # Power Generated by System
      - name: "System Power Generated"
        unique_id: "system_power_generated"
        unit_of_measurement: "kW"
        state_class: "measurement"
        device_class: "power"
        state: >-
          {% set solar_value = states('sensor.gosungrow_virtual_1134569_1_1_1_p24') | float(0) / 1000 %}
          {{ solar_value | round(2) }}

      # Energy Sent Out
      - name: "Exported Energy"
        unique_id: "exported_energy"
        unit_of_measurement: "kWh"
        state_class: "total_increasing"
        device_class: "energy"
        state: >-
          {% set grid_feed_value = states('sensor.gosungrow_virtual_1134569_7_1_1_p8018') | float(0) %}
          {{ abs(grid_feed_value) if grid_feed_value < 0 else 0 }}

      # Consumed Energy
      - name: "Energy Used Locally"
        unique_id: "energy_used_locally"
        unit_of_measurement: "kWh"
        state_class: "total_increasing"
        device_class: "energy"
        state: >-
          {% set solar_generated = states('sensor.gosungrow_virtual_1134569_1_1_1_p24') | float(0) %}
          {% set exported = states('sensor.exported_energy') | float(0) %}
          {{ solar_generated - exported if solar_generated >= exported else 0 }}

# MQTT Integration for Sharing Data (Optional)
mqtt:
  sensor:
    - name: "Shared Power Data"
      unique_id: "shared_power_data"
      state_topic: "system/power/summary"
      value_template: >-
        {
          "generated": "{{ states('sensor.system_power_generated') }}",
          "exported": "{{ states('sensor.exported_energy') }}",
          "local_use": "{{ states('sensor.energy_used_locally') }}"
        }
      json_attributes_topic: "system/power/summary"

bamcauliffe, 2025-04-09:
Keensy,
The live feedback entities that I am using are
Solar generating = sensor.gosungrow_getpslist_devices_xxxxxxxx_curr_power (positive only number)
Electricity purchasing = sensor.gosungrow_virtual_xxxxxxxxx_p8018 (positive number for buying electricity and negative number for exporting electricity). This is what you are using for "Energy Sent Out"
Home Load = sensor.gosungrow_getpsdetail_1202522_p83106

keensy, 2025-04-09:
> Keensy,
> The live feedback entities that I am using are
> Solar generating = sensor.gosungrow_getpslist_devices_xxxxxxxx_curr_power (positive only number)
> Electricity purchasing = [...]

Thanks mate I ended up working this out. Thanks for the response

bamcauliffe, 2025-04-09:
Keensy,
Great to hear that you worked out the answer to your problem.
If you are happy to share, it would be good to hear what you worked our in the end.
If you are happy that you have resolved the problem and happy with the outcome, is this issue good to be closed?
Does the raiser close issues or the owner of the integration?
I have not closed an issue yet.