MickMake/GoSungrow #125: Getting er_invalid_appkey repeatedly, new installation

NQTrojan, 2024-12-21:
Hi All
Would appreciate some assistance with trying to resolve this issue, has been happening since I've installed the addin.  I've just restarted the Home Assistant server after it being offline for a while
Troubleshooting I've done
checked access to the host: augateway.isolarcloud.com
Restarted the Server and the Host VM
Other Integrations are working fine
Checked my login details, although this appears to be an API Key issue

Do I have to check for a new API key somewhere?

Any assistance would be greatly appreciated.

**Error in logs**
[06:07:03] INFO: Login to iSolarCloud using gateway https://augateway.isolarcloud.com ...
Error: appkey is incorrect 'er_invalid_appkey'

**Complete log with my details removed**
[06:07:02] INFO: Setting up GoSungrow config ...
[06:07:03] INFO: Writing GoSungrow config ...
Using config file '/data/.GoSungrow/config.json'
New config:
+-------------------+------------+---------------------------+--------------------------------+-----------------------------------+
|       FLAG        | SHORT FLAG |        ENVIRONMENT        |          DESCRIPTION           |        VALUE (* = DEFAULT)        |
+-------------------+------------+---------------------------+--------------------------------+-----------------------------------+
| --config          |            | GOSUNGROW_CONFIG          | GoSungrow: config file.        | /data/.GoSungrow/config.json      |
| --debug           |            | GOSUNGROW_DEBUG           | GoSungrow: Debug mode.         | true                              |
| --quiet           |            | GOSUNGROW_QUIET           | GoSungrow: Silence all         | false *                           |
|                   |            |                           | messages.                      |                                   |
| --timeout         |            | GOSUNGROW_TIMEOUT         | Web timeout.                   | 1m0s                              |
| --user            | -u         | GOSUNGROW_USER            | SunGrow: api username.         |             |
| --password        | -p         | GOSUNGROW_PASSWORD        | SunGrow: api password.         |             |
| --appkey          |            | GOSUNGROW_APPKEY          | SunGrow: api application key.  | 93D72E60331ABDCDC7B39ADC2D1F32B3  |
|                   |            |                           |                                | *                                 |
| --host            |            | GOSUNGROW_HOST            | SunGrow: Provider API URL.     | https://augateway.isolarcloud.com |
|                   |            |                           |                                | *                                 |
| --token-expiry    |            | GOSUNGROW_TOKEN_EXPIRY    | SunGrow: last login.           | 2023-05-22T19:55:05               |
| --save            | -s         | GOSUNGROW_SAVE            | Save output as a file.         | false *                           |
| --dir             |            | GOSUNGROW_DIR             | Save output base directory.    |  *                                |
| --mqtt-user       |            | GOSUNGROW_MQTT_USER       | HASSIO: mqtt username.         |                             |
| --mqtt-password   |            | GOSUNGROW_MQTT_PASSWORD   | HASSIO: mqtt password.         |                          |
| --mqtt-host       |            | GOSUNGROW_MQTT_HOST       | HASSIO: mqtt host.             | core-mosquitto                    |
| --mqtt-port       |            | GOSUNGROW_MQTT_PORT       | HASSIO: mqtt port.             |                              1883 |
| --modbus-user     |            | GOSUNGROW_MODBUS_USER     | Modbus username.               |  *                                |
| --modbus-password |            | GOSUNGROW_MODBUS_PASSWORD | Modbus password.               |  *                                |
| --modbus-host     |            | GOSUNGROW_MODBUS_HOST     | Modbus host.                   |  *                                |
| --modbus-port     |            | GOSUNGROW_MODBUS_PORT     | Modbus port.                   | 502 *                             |
+-------------------+------------+---------------------------+--------------------------------+-----------------------------------+
[06:07:03] INFO: Login to iSolarCloud using gateway https://augateway.isolarcloud.com ...
Error: appkey is incorrect 'er_invalid_appkey'
Usage:
  GoSungrow api login [flags]
Examples:
	GoSungrow api login  
Flags: Use "GoSungrow help flags" for more info.
Additional help topics:
ERROR: appkey is incorrect 'er_invalid_appkey'
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

Paraphraser, 2024-12-21:
Please see [this gist](https://gist.github.com/Paraphraser/cad3b0aa6428c58ee87bc835ac12ed37) which deals with several scenarios for running GoSungrow. It sounds like you will need to follow the steps in Part 2 of the gist. Each "part" winds up at the configuration instructions which includes info about the app key.

NQTrojan, 2024-12-21:
Hi Paraphraser
Thank you for pointing me to this resource, I appear to have been successful in getting through that error by trying another app key in that document.

I ran into some more errors due to other issues with the Home Assistant Server which I've been able to update and resolve and I appear to be making some progress but am stuck at this point

I am now getting this error after it appears to login to both iSolarCloud and my Home Assistant instance correctly

Error: network Error : non-CONNACK first packet received

Again thank you for your assistance with the earlier

**Full Log**
[11:53:57] INFO: Login to iSolarCloud using gateway https://augateway.isolarcloud.com ...
Email:	deleted
Create Date:	Fri May 05 10:12:31 CST 2023
Login Last Date:	2024-12-21 19:44:36
Login Last IP:	
Login State:	1
User Account:	deleted
User Id:	deleted
User Name:	deleted
Is Online:	false
Token:	395675_02606f6671944c14b7c60c9f7fc9dc25
Token File:	/data/.GoSungrow/AppService_login.json
Email:	deleted
Create Date:	Fri May 05 10:12:31 CST 2023
Login Last Date:	2024-12-21 19:53:57
Login Last IP:	
Login State:	1
User Account:	deleted
User Id:	deleted
User Name:	deleted
Is Online:	false
Token:	395675_a1b763cf7d8448ab9e871129bc1349ef
Token File:	/data/.GoSungrow/AppService_login.json
[11:53:57] INFO: Syncing data from gateway https://augateway.isolarcloud.com ...
Email:	
Create Date:	Fri May 05 10:12:31 CST 2023
Login Last Date:	2024-12-21 19:53:57
Login Last IP:	
Login State:	1
User Account:	deleted
User Id:	deleted
User Name:	deleted
Is Online:	false
Token:	395675_a1b763cf7d8448ab9e871129bc1349ef
Token File:	/data/.GoSungrow/AppService_login.json
2024/12/21 11:53:57 INFO: Connecting to MQTT HASSIO Service...
2024/12/21 11:53:57 INFO: Connecting to SunGrow...
2024/12/21 11:53:58 INFO: Found SunGrow 5 devices
Error: network Error : non-CONNACK first packet received
Usage:
  GoSungrow mqtt run [flags]
Aliases:
  run, 
Examples:
	GoSungrow mqtt run  
Flags: Use "GoSungrow help flags" for more info.
Additional help topics:
ERROR: network Error : non-CONNACK first packet received
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

Paraphraser, 2024-12-21:
Can't be 100% sure but one possible explanation is that this error is talking about TCP handshaking where "connections" are established via packets with the SYN flag set, then SYN+ACK, then ACK. Then, as the client and server exchange packets, you expect to see ACK (ie "acknowledgement") packets at regular intervals confirming reception.

It *might* mean that something other than an ACK came back when an ACK was expected.

That's all at layer four of the OSI model.

Alternatively, the layer 7 protocol (aka "the API") between GoSungrow and iSolarCloud might have its own definition of ACK and the error might be talking about something missing or out-of-sequence at that level.

All I can really say is:

1. It isn't happening to me (Home Assistant running as a guest on Proxmox-VE) so it probably isn't either *systemic* or a temporary glitch affecting the AU gateway (that's my gateway too).
2. I have a bog standard Netcomm router from my ISP (AussieBroadband) and the only firewalling occurring is whatever is baked into that. I don't have device-level firewalls enabled anywhere.

I mention firewalls because any time packets go missing a misconfigured firewall is a likely suspect. Go anything that fits the bill?

NQTrojan, 2024-12-21:
Hi
Nothing stands out, can I please check what API Key you are using?

Regards, Glenn

Paraphraser, 2024-12-21:
"B0455FBE7AA0328DB57B59AA729F05D8" but previous experience (other issues) suggests there's no relationship between app key and gateway server. Some people reported using this "B0" key then having to switch to the "ANDROID" key. There was a theory the switch of appkey might be triggered by logging in via the web GUI but I tried that and wasn't forced to switch.

🤷‍♂️

NQTrojan, 2024-12-29:
Hey Phil
Just letting you know I had another look at this and found the issue.
I had mistakenly assumed that the MQTT login was my Home Assistant login
which of course is not true.  I have a better understanding of HA now and
recognised the issue straight away.

After resetting those MQTT settings and using the Android key from the
article you pointed me to I can now confirm it's all working beautifully.

Thank you so much for your assistance and patience, really appreciate it.