MickMake/GoSungrow #126: Limit the power supply

apassi, 2025-02-20:
Hi,

Is it possible to adjust the amount the power the inverter feeds to grid with this jewel? I know it is possible to adjust with local admin access, but i think it is not possible throught the sungrow web ui.
I found "P-limit-actual" point_id under the docs.

- Arto

Paraphraser, 2025-02-20:
If you ever get an answer to this question, **please** post it here. My previous inverter was a SolaX and I could set feed-in limit at anything between zero and the rating of the inverter. I can't think of any reason why it shouldn't be possible with Sungrow. It's darned annoying that it's not documented anywhere.

Seems to me it's your inverter so it should be your choice as to whether you want to feed-in surplus. Particularly if they're going to start charging for the privilege - on top of piddling rates. Serious insult. If they don't want it, fine, don't take it. No skin off my nose.

apassi, 2025-02-26:
@Paraphraser i was able find a way to change the limit with programatically, here some discussion about it:
https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/discussions/393#discussioncomment-11237099

and i just upload some bash script, which uses bash, curl, jq, uwsc, and expect to login, gain the token, and change the limit value:
https://github.com/apassi/sg-power-limit

Paraphraser, 2025-02-26:
@apassi sorry for the slow reply - I've been distracted.

> You know that it is possible to change the feed-in limit though sungrow android app, with local admin access?

No, I did not know that. My previous inverter was a SolaX. It was a bit temperamental so, of necessity, I had to learn how to get down and dirty with its parameters.

I have never really done more with iSolarCloud than one or two screens in. I'm using the iPad app. I have no idea whether the iPad and Android versions are the same. Even though I've had the SG5.0RS for two years, I'm still at the newbie end of the learning curve. My focus at the beginning was on downloading "yesterday's" data to continue a time-series that began with the SolaX. I use GoSungrow for that, triggered by a cron job. Once I got that working I was content to leave things be.

The only problem I have ever had with the SG5.0RS was quite recent when I noticed that the front panel "kWh Today" wasn't resetting to zero each day. The stats coming from GoSungrow weren't affected. I power-cycled the inverter and the problem went away. I put it down to a side effect of thunderstorms which had produced some short outages and brownouts.

Anyway, are you talking about Support, Commissioning Tools, Local Access? That's the only thing I can find mentioning local access. Seems to want me to scan the QR code on the WiFi dongle. The "commissioning" bit troubles me - the inverter is already commissioned.

I'll take a look at your other links. Thanks.