MickMake/GoSungrow #131: Can't find desired data to export

swedishstyle, 2025-05-28:
All I'm after with this program is a simple CSV export of the following from my inverter:

Total Active Power (kW)
Yield Today (kWh)
Total yield (kWh)
Ambient Temperature (C)

For any given day, with 5 min increments in the data. I searched for hours through examples and tried to find it on my own, but had absolutely no luck at all. I can connect and get data just fine, but not the data I'm after. Am I just thick, or am I asking for data outside of what the API offers? I can see all this data (except for the ambient temp, that's not crucial, but a nice to have) in the web portal, under Device -> Device List -> My Inverter -> Curve.

I can manually export those CSV's, but obviously that's not ideal.

I realise the maintainer has gone silent, so thanks in advance to any kind souls who would take some time to show me what I'm missing!

swedishstyle, 2025-05-29:
Thanks - getting through it now, but Im hitting a block with the date grabbers:
```
FETCH_DATE=$(date -v -${BIAS}d "+%Y%m%d")
date: invalid option -- 'v'
Try 'date --help' for more information.
```

I've tried running without the -v, but then I get:
```
date: invalid option -- '1'
Try 'date --help' for more information.
```

Not sure what the -v is supposed to be achieving, I can't find doco on that anywhere?

swedishstyle, 2025-05-29:
Your wild stab would be correct, I'm trying this on Ubuntu server :)

Thanks for the detailed response, I really appreciate it! However I discovered a line in the GoSungrow docs that I hadn't noticed before, queryMultiPointDataList that outputs in json, so I've got that working and wrote some python to get the data in my db. Now it's working fine, so I don't think I need to continue with your project, but thanks for leading me in the right direction! 

I won't share the script as I'm using sql server and it's a bit custom, but once you have json it shouldn't be too hard to get it wherever you want it from there.

closed 2025-05-29 by swedishstyle

Paraphraser, 2025-05-29:
I had not noticed `queryMultiPointDataList`. Thanks.

In my own situation, I had migrated from a SolaX inverter to a Sungrow. The SolaX Cloud offered CSV directly (and nothing else, as far as I could tell). I had already written an "extract+transform+load" tool circa 2009 to load CSV into SQLite, so it was the hammer I had to hit the nail in front of me when the SolaX came along in 2015.  The trickiest bit, at least for me, was figuring out how to set up authentication cookies for the download via `curl`. In 2023 with the Sungrow, there may have been some way of obtaining CSV from iSolarCloud via `curl` but it wasn't immediately obvious. GoSungrow would at least fetch the data points. I recall seeing mentions of an option for CSV output but nothing I tried worked and I didn't have either the time or inclination to wade through the Go code to figure out why. But I was getting regular text, so I could solve the "I need CSV at this point" with a hack, and feed that to the originally-designed-for-SolaX toolchain, thereby continuing the time series. Truth to tell, if I had a do-over, I'd probably bolt it into the MING stack (Mosquitto, InfluxDB, Node-RED, Grafana) I use for IoT devices.

I'm glad you solved your problem.

Paraphraser, 2025-05-30:
I know you said you decided to go another way but, in case anyone else was following this discussion, I've made the requisite [changes](https://github.com/Paraphraser/gosungrow-fetch/commit/d1970dd9de62034f801ae5a04a8ff18b9879527b) to the script so that `date` will do the correct calculations on Linux as well as macOS.