MickMake/GoSungrow #134: server misbehaving

kavo91, 2025-06-30:
Getting following error, was previously working. Only noticed that values stopped updating:

`[04:13:57] INFO: Login to iSolarCloud using gateway https://augateway.isolarcloud.com ...
Error: Post "https://augateway.isolarcloud.com/v1/userService/login": dial tcp: lookup augateway.isolarcloud.com on 127.0.0.11:53: server misbehaving
Usage:
  GoSungrow api login [flags]
Examples:
	GoSungrow api login  
Flags: Use "GoSungrow help flags" for more info.
Additional help topics:
ERROR: Post "https://augateway.isolarcloud.com/v1/userService/login": dial tcp: lookup augateway.isolarcloud.com on 127.0.0.11:53: server misbehaving
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped`

Paraphraser, 2025-06-30:
**HOW** are you using GoSungrow? Home Assistant add-on or separate Docker container?

The mentions of 127.0.0.11 suggest a DNS problem. 

Have you stopped then started the addon/container? Not just restart. Actual stop/start to force the container to be recreated from its image.

Are you sure you are using the patched container as per part 2 (HA) or part 4 (container) of [this gist](https://gist.github.com/Paraphraser/cad3b0aa6428c58ee87bc835ac12ed37)? I'm trying to rule out unintended regression.

What happens if you paste `https://augateway.isolarcloud.com/` into your browser? Do you get the word "success"? 

What happens if you SSH to your Home Assistant instance (or SSH to the host where the container is running) and run the command:

```
getent hosts augateway.isolarcloud.com
```

Silence (indicating a DNS problem) or an answer like the following which indicates success:

```
13.237.163.79   au-prod-cloudservice-gateway80-1681975692.ap-southeast-2.elb.amazonaws.com augateway.isolarcloud.com
```

Or something that looks like an error? On HA I get one line as above. On Debian I get three lines.

I use the au gateway and it's all working for with no signs of recent interruptions me so my guess is this is less likely to be systemic and more likely to be something unique to your setup and path to the Internet.

kavo91, 2025-06-30:
I was using it on HAOS with the original install. Was working perfectly until a few days ago. I had successful SSH commands. I just updated to the part 2 patched container from the gist and appears to be working