MickMake/GoSungrow #136: One more time: Found SunGrow 0 devices

WhiteLion13, 2025-07-10:
Token File:	/data/.GoSungrow/AppService_login.json
[02:38:46] INFO: Syncing data from gateway https://gateway.isolarcloud.eu ...
2025/07/10 02:38:46 INFO: Connecting to MQTT HASSIO Service...
2025/07/10 02:38:46 INFO: Connecting to SunGrow...
2025/07/10 02:38:48 INFO: Found SunGrow 0 devices
2025/07/10 02:38:48 INFO: Option[fetchschedule] set to '2m'
2025/07/10 02:38:48 INFO: Caching Sungrow metadata...
2025/07/10 02:38:48 INFO: Cached 0 Sungrow data points...
No results found.
2025/07/10 02:38:48 INFO: Syncing 1 entries with HASSIO from getPsList.
CU
2025/07/10 02:38:48 INFO: Starting ticker...
2025/07/10 02:38:48 INFO: Fetch Schedule: 2m
2025/07/10 02:38:48 INFO: Sleep Delay:    40s
2025/07/10 02:41:48 INFO: Syncing 1 entries with HASSIO from getPsList.
U
No results found.
————————————————————————

Any solution for this issue?
The previous users with this problem solved it, but never said how they did it.  :-(
Thanks for any suggestions or help!
Chris

Paraphraser, 2025-07-11:
Sounds like it could be a valid account but your equipment isn't actually logging to that account.

You *might* be able to get more information like this (I'm assuming you have the "Advanced SSH & Web Terminal" installed in your Home Assistant instance):

1. Use SSH to connect to Home Assistant.
2. Open a shell into the GoSungrow add-on container:

	```
	docker exec -it addon_ba22da74_gosungrow bash
	```

	If that doesn't work then run:

	```
	docker ps -a --format "table {{.Names}}\t{{.RunningFor}}\t{{.Status}}\t{{.Size}}" --filter name=gosungrow
	```

	which will give you the correct name in the NAMES column. Replace `addon_ba22da74_gosungrow` in the previous command with the correct name.

3. Now that you have a shell inside the container, try logging-in to iSolarCloud:

	```
	GoSungrow --config /data/.GoSungrow/config.json api login
	```
	
	If login fails then you might get a hint as to why from the output.

4. If login succeeds then run:

	```
	GoSungrow --config /data/.GoSungrow/config.json show ps list
	```
	
	If you get nothing then it probably confirms my original suspicion.
	
5. You need two `exit` commands (or two <kbd>control</kbd>+<kbd>d</kbd>) to drop out of the container and then disconnect from SSH.

Note that you always need:

```
--config /data/.GoSungrow/config.json
```

when working inside the container. That file gets seeded from the "configuration" tab for the GoSungrow add-on in Home Assistant.

WhiteLion13, 2025-07-13:
➜  ~ docker exec -it addon_ba22da74_gosungrow bash                                                             
bash-5.1# GoSungrow --config /data/.GoSungrow/config.json api login
Email:  chriXXXXXXXXXXXXXXXXXXXXX
Create Date:    Thu Jul 10 05:24:18 CST 2025
Login Last Date:        2025-07-14 02:57:32
Login Last IP:
Login State:    1
User Account:   WhXXXXXXXXXXXXXX
User Id:        71XXXXXXX
User Name:      CS_Vollzugriff/Installateur
Is Online:      false
Token:  710117_69c127b60ba04947bf58cb2729e87e47
Token File:     /root/.GoSungrow/AppService_login.json
bash-5.1# GoSungrow --config /data/.GoSungrow/config.json show ps list
┏━━━━━━━━┳━━━━━━━┳━━━━━━━━━━━━━┳━━━━━━━━━━━━━┳━━━━━━━━━━━━┳━━━━━━━━━━┳━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━┓
┃ Ps Key ┃ Ps Id ┃ Device Type ┃ Device Code ┃ Channel Id ┃ Serial # ┃ Factory Name ┃ Device Model ┃
┣━━━━━━━━╇━━━━━━━╇━━━━━━━━━━━━━╇━━━━━━━━━━━━━╇━━━━━━━━━━━━╇━━━━━━━━━━╇━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━┫
┗━━━━━━━━┷━━━━━━━┷━━━━━━━━━━━━━┷━━━━━━━━━━━━━┷━━━━━━━━━━━━┷━━━━━━━━━━┷━━━━━━━━━━━━━━┷━━━━━━━━━━━━━━┛

bash-5.1# exit

Thank you for trying to help.
This is the result of your "instructions". 
BTW, I didn´t mention the type of my inverter: SunGrow SH20T with the WiNet-S2-dongle, which is connected via LAN.

The thing is, when I login on the official iSolarCloud website, I can see all the data. So my equipment does connect and log to the account. Any other ideas? Highly appreciated.  :-)

Thanks from Austria,
Chris

Paraphraser, 2025-07-14:
I'm no expert on any of this. But, when I compare your output with what I see when I login using GoSungrow, the one thing that stands out is:

```
User Name: CS_Vollzugriff/Installateur
```

whereas mine is:

```
User Name: Phill
```

As far as I can tell, "Phill" comes from the "Nickname" field in:

* iSolarCloud » Account » tap in the Profile at the top of the screen

I'm pretty sure I would have typed "Phill" into some setup screen at some point.

When I ask for a translation of "Vollzugriff Installateur" I get "full access installer". That at least *suggests* the possibility that you are using an account which is intended to give you access to a whole lot of different customers' equipment. If so then, logically, there must be some step that goes *before* `show ps list` where you have to identify the customer so that the API knows which customer's `ps list` to fetch.

I have no idea whether MickMake is just an ordinary user or an installer or had some other level of privilege which allowed him to see more than his own equipment. If I assume MickMake is an "ordinary user" who can only see his own equipment, then he may not even have been aware that there might be situations where it would be necessary to "identify the customer", either as a preliminary step or by passing another parameter on API calls. And if *that's* true then it would probably imply that the necessary functionality doesn't exist in GoSungrow. Still, this is all speculation and, as I said at the start, I'm no expert.  

<hr>

So let's rule a line under that. What I do recall is when my own installer started to set things up, he was using a mixture of his own credentials and my email address. I had read a document on one of SunGrow's web sites which said it was something a user should do. Plus, he was using the wrong email address so I stopped him and did it myself. I know he can see my plant and I presume he can see the plants of all the other customers he has installed. I, of course, can only see my own plant.

My **guess** is that, when it comes to **your** plant that you probably need to login as an ordinary user rather than as a "Vollzugriff Installateur". That would avoid the "pre-identify the customer" step.

To be clear, when I say "login as an ordinary user" I mean the values in the `sungrow_user` and `sungrow_password` fields of the Configuration tab of the GoSungrow add-on.

At the command-line, the equivalent would be something like:

```
cp /data/.GoSungrow/config.json .
GoSungrow --config config.json config write --user «ordinaryUsername» --password «ordinaryUserPassword»
GoSungrow --config config.json api login
GoSungrow --config config.json show ps list
```

In words:

1. Make a copy of the running config file.
2. Set credentials in the copy.
3. Login using those credentials.
4. See if you can fetch the PS list.

I'd advise caution at step 3 because it looks to me like iSolarCloud imposes a lockout if you get too many login failures in a row.

<hr>

You said you can see your own data when you login to the iSolarCloud web site. I don't think that's surprising because, presumably, you are both the installer and the customer so, when you login as an installer, you can see all your customers including yourself. If you need to click on your own plant to see your own data, it's that "click on your own plant" step that I'm talking about which is probably missing from GoSungrow. If you see your own plant by default then, presumably, there's some "settings" or "profile" somewhere via which the web site knows to show you your own plant first. Still just guessing...

WhiteLion13, 2025-07-14:
Good morning!

Thanks a lot for suggestions. Unfortunately, it makes no difference if I login with my "regular end-user" credentials or with the credentials as an installer. I got the same result ("found 0 Sungrow devices") before when I used my regular credentials. I set up another accout as an installer after that just to try and see if I get better results with a "full-access"-account like an installer. BUT - both of them return "0 Sungrow devices".  :-(

Paraphraser, 2025-07-15:
Once again I feel the need to begin by emphasising that I am not an expert in any of this. I have a lot of experience writing code in many programming languages but Go has never been among them. This is just guesswork.

If we take the stack-trace at face value and assume the line numbers are accurate then the last line of this:

```
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0xc8 pc=0x8dad53]
goroutine 1 [running]:
github.com/MickMake/GoSungrow/iSolarCloud/AppService/queryDeviceList.(*EndPoint).SetBatteryPoints(, {{, , }}, {0xc000a92c00, {0xc000885897, 0x7}, {0xc21553309baaaaf6, 0x46cf303, 0x4b5fa20}, ...})
/Users/isit/git/GoSungrow/iSolarCloud/AppService/queryDeviceList/data.go:336 +0x4d3
```

leads us to [this](https://github.com/MickMake/GoSungrow/blob/391253aaadd2cae9df32c95bec9e6b9bbf83f4d6/iSolarCloud/AppService/queryDeviceList/data.go#L336) line in the code:

```
batteryChargeEnergy.DataStructure.PointUpdateFreq = GoStruct.UpdateFreqDay
```

That's an assignment. The right hand side is initialised [here](https://github.com/MickMake/GoSungrow/blob/391253aaadd2cae9df32c95bec9e6b9bbf83f4d6/iSolarCloud/api/GoStruct/const.go#L43) and an equivalent assignment occurs at line 332 in `SetBatteryPoints()` so the right hand side would not appear to be the cause of the error.

That implies the left hand side is the problem. The `batteryChargeEnergy` variable appears to be initialised at line 334:

```
batteryChargeEnergy := entries.CopyPointFromName(epp.AddString("p13174"), epp, "battery_charge_energy", "Battery Charge Energy (p13174)")
```

The crash ("invalid memory address or nil pointer dereference") suggests that the most likely reason is `entries.CopyPointFromName()` failed.

Lines 330..332 have the same pattern and apparently succeeded. The most plausible explanation for the difference in behaviour is that the `p13029` point exists while the `p13174` point does not.

I find support for this in the structure of the [`CopyPointFromName()`](https://github.com/MickMake/GoSungrow/blob/391253aaadd2cae9df32c95bec9e6b9bbf83f4d6/iSolarCloud/api/struct_data.go#L154-L175) function. The `Current` variable is *declared* but is only *initialised* if one of the three `if` tests succeeds, which means there's a path through the code where `Current` is returned uninitialised (and, given the way most memory-safe languages behave these days, is probably nil).

> Were this **my** code, I'd at least consider initialising `Current` to a valid structure with sensible defaults which would stick out like a sore thumb in reports or MQTT messages coming out of GoSungrow. But whether that would make sense in this context is something I can't say. Maybe allowing the code to crash is appropriate.

The main reason I'm laying out this thought process in detail is in the hope people who know more Go than I do can chime in if I've misunderstood.

I get silence when I run this command:

```
$ GoSungrow show ps points | grep -e "p13029" -e "p13174"
```

Silence means that neither of those point identifies is present in my list. One explanation is I simply do not have batteries so battery-related metrics do not make it into my list.

On the other hand:

```
$ GoSungrow show ps points | grep -i -c -e "battery" -e "charge"
64
```

A count of 64 means a significant number of rows that include those words.

I'm hoping that if you run:

```
$ GoSungrow show ps points | grep -e "p13029" -e "p13174"
```

that you'll get a hit on `p13029` but not `p13174`.

It might also be instructive to run this:

```
$ GoSungrow show ps points | grep -e "p13003" -e "p13029" -e "p13112" -e "p13116" -e "p13119" -e "p13121" -e "p13122" -e "p13126" -e "p13130" -e "p13134" -e "p13144" -e "p13147" -e "p13149" -e "p13150" -e "p13173" -e "p13174" -e "p13199" -e "p8354"
```

Those are all the point identifiers mentioned in the `queryDeviceList` package.

JerryWho, 2025-07-19:
I found https://github.com/MickMake/GoSungrow/issues/18#issuecomment-1445300680 So it seems to me, Sungrow is messing up with the rights. Management rights don’t seem to be enough 😳

Paraphraser, 2025-07-21:
@JerryWho please see [comment on issue 70](https://github.com/MickMake/GoSungrow/issues/70#issuecomment-3094860356) and see whether that makes sense in your context.

My working theory is:

1. Use either iSolarCloud app or a web browser.
2. Login as the **owner**.
3. Configure GoSungrow to also login as the **owner**.

I haven't heard the term "Management rights" until now. Can you elaborate?